kactus2/kactus2dev #24: Nested Vendor Extensions are silently deleted

DanChianucci, 2019-03-14:
If the model contanis vendor extensions which are nested hierarchically, only the top level element is saved.  It looks like this is a new bug, as previous versions did not do this.

For Example if I open and save the following,
```
<?xml version="1.0" encoding="UTF-8"?>
<ipxact:component xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:ipxact="http://www.accellera.org/XMLSchema/IPXACT/1685-2014" xmlns:kactus2="http://kactus2.cs.tut.fi" xmlns:my_vendor="http://www.my_vendor.com">
  <ipxact:vendor>my_vendor</ipxact:vendor>
  <ipxact:library>my_library</ipxact:library>
  <ipxact:name>my_name</ipxact:name>
  <ipxact:version>my_version</ipxact:version>
  <ipxact:vendorExtensions>
    <my_vendor:parent>
      <my_vendor:child_0>Some Value</my_vendor:child_0>
      <my_vendor:child_1>
        <my_vendor:child_1_0>Some Value</my_vendor:child_1_0>
        <my_vendor:child_1_1>Some Value</my_vendor:child_1_1>
      </my_vendor:child_1>
      <my_vendor:child_2>
        <my_vendor:child_2_0>Some Value</my_vendor:child_2_0>
        <my_vendor:child_2_1>
          <my_vendor:child_2_1_0> Some Value</my_vendor:child_2_3_0>
        </my_vendor:child_2_1>
      </my_vendor:child_2>
    </my_vendor:parent>
  </ipxact:vendorExtensions>
</ipxact:component>
```

 the saved copy will only contain
```
<?xml version="1.0" encoding="UTF-8"?>
<ipxact:component xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:ipxact="http://www.accellera.org/XMLSchema/IPXACT/1685-2014" xmlns:kactus2="http://kactus2.cs.tut.fi" xmlns:my_vendor="http://www.my_vendor.com" xsi:schemaLocation="http://www.accellera.org/XMLSchema/IPXACT/1685-2014 http://www.accellera.org/XMLSchema/IPXACT/1685-2014/index.xsd">
	<ipxact:vendor>my_vendor</ipxact:vendor>
	<ipxact:library>my_library</ipxact:library>
	<ipxact:name>my_name</ipxact:name>
	<ipxact:version>my_version</ipxact:version>
	<ipxact:vendorExtensions>
		<my_vendor:parent>
			<my_vendor:child_1>
				<my_vendor:child_1_1/>
			</my_vendor:child_1>
			<my_vendor:child_2>
				<my_vendor:child_2_1/>
			</my_vendor:child_2>
		</my_vendor:parent>
	</ipxact:vendorExtensions>
</ipxact:component>
```

It seems to lose top level children if they only contain text.
Keep top level children if they themselve contain children.
It removes the content from gradchildren nodes, and only keeps the last element.

epekkar, 2019-03-15:
Thank you, Dan.

We are currently working on the Vendor Extension editor and nested extensions is one of the features that will be improved soon.

epekkar, 2019-04-01:
Fixed in a6a227d14e274be50b7f1bed395be01103a914bc.

closed 2019-04-01 by epekkar

DanChianucci, 2019-04-01:
Awesome, Thanks!