kactus2/kactus2dev #31: Kactus doesn't correctly deal with XML namespaces

andreasWallner, 2019-08-31:
I'm trying to use Kactus with a bunch of scripts/tools that we've used for a while. I encountered a problem while doing that with the XML namespace handling in Kacuts. Our tools use an XML libraries that do not explicitly place the namespace on every node (and can't be forced to do so) but use the 'xmlns' attribute to set the namespace on the root node of the file. Those files can't be worked on using Kactus because the deserialization assumes the namespace prefix 'ipxact:' to be present on all tags.
I'd be willing to provide a pull request to make the deserialization standards conformant, but wanted to ask first if such a change would be welcomed at all.

epekkar, 2019-09-03:
Hi Andreas,
Your suggested contribution would be warmly welcome. I think two questions need to be considered: 

1. Is your need only for reading XML into Kactus2 or do you require the XML files created by Kactus2 to follow the same rule? The former I see as straightforward (yet requires effort) to implement, but the latter could be problematic.  

2. Do you need to consider vendor extensions (in their respective namespaces) as well?

andreasWallner, 2019-09-03:
Hi,
ad 1) Import alone should be sufficient. The files that you are currently generating are valid and correct XMLs as far as I've seen. Our tools should therefore understand them w/o problems. Something for me internally is to find a solution that we do not bounce between serializations in our version control, but that's a different topic.

ad 2) I need the import to at least preserve the namespace of vendor extensions - such that a saved file will represent the same data as on import.

Regarding vendor extensions: as far as I've seen you currently do not show them in the UI, is that correct?

I'll start with the modifications somewhen close to the WE, but I guess it will take me a bit as I'm doing that on the side.

epekkar, 2019-09-03:
We recently introduced a prototype editor for vendor extensions, so at least you can check, if they import as expected. The editor can currenlty only show/edit component (top-level) and port extensions. You can enable it on the toolbar View->Visible Windows->Vendor Extensions.