kactus2/kactus2dev #32: Clocks values are limited to 99.99

suzizecat, 2019-10-04:
Hello,

In the Other Clocks editor, all numeric values are limited in edition to 99.99.
Coupled with the fact that it is not possible to specify the timings in anything aside nanoseconds and picoseconds, it (artificially) forbid the support for clock over 100ns or below 0.01ps.

It is possible to work around that using right-click > paste but this isn't practical.

Could this limitation be lifted and other time units added (at least up to the second) within the time unit drop-down ?
For me, being able to work with at least a 999.999 time format (and more time units)  would be great. 

Best regards,
Julien FAUCHER

------

I'm working with the 3.7.0 version compiled under Ubuntu 19.04

epekkar, 2019-10-07:
Hi suzizecat,

Thanks for reporting the issue. We'll fix it asap.

teuhom, 2019-10-07:
Fixed in 57a825a81e7c721b93a12e615b8af5c09cd1af7e

closed 2019-10-07 by teuhom