kactus2/kactus2dev #42: Kactus2 3.9 windows does not start

0tti, 2021-09-10:
Windows installer.

Error message:
The program can't start because python38.dll is missing from your computer. Try reinstalling the program to fix this problem.

0tti, 2021-09-10:
Downloaded the dll and put it into the installation folder. The error above disappears but new errors show up:

QVariant::load: unknown user type with name HighlightStyleDesc.
QVariant::load: unknown user type with name HighlightStyleDesc.
QVariant::load: unknown user type with name HighlightStyleDesc.
QVariant::load: unknown user type with name HighlightStyleDesc.
QVariant::load: unknown user type with name HighlightStyleDesc.
QVariant::load: unknown user type with name HighlightStyleDesc.
QVariant::load: unknown user type with name HighlightStyleDesc.
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = 'Kactus2.exe'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = 'C:\\Program Files\\Kactus2\\Kactus2.exe'
  sys.base_prefix = ''
  sys.base_exec_prefix = ''
  sys.executable = 'C:\\Program Files\\Kactus2\\Kactus2.exe'
  sys.prefix = ''
  sys.exec_prefix = ''
  sys.path = [
    'C:\\Program Files\\Kactus2\\python38.zip',
    '.\\DLLs',
    '.\\lib',
    'C:\\Program Files\\Kactus2',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00002304 (most recent call first):
<no Python frame>

epekkar, 2021-09-10:
I'll add the DLL in the installer. 
You can ignore the warning `QVariant::load: unknown user type with name HighlightStyleDesc.`.

0tti, 2021-09-10:
Maybe I was not clear, but Kactus2 does **not** start even with the dll in place. There is more missing.

Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

epekkar, 2021-09-13:
It seems your Python installation is somehow faulty. Can you run Python interpreter or other scripts normally? Please check that and re-install Python if need be.

0tti, 2021-09-13:
I have a python 3.6.8 installed. Works fine.
Kactus2 3.6 and 3.8 worked also without any python installation in the past. 
 
Removed the local python 3.6.8 from path environment variable. 

- Kactus2 3.9 still produce the error from above.
- Kactus2 3.6 starts as usuall

Is there some further dll missing which includes the 'encodings' module?
You may try an installation on an machine without python installed.

epekkar, 2021-09-13:
Earlier versions of Kactus2 (<3.9.0) did not require Python at all, we added that in the latest version. 

Module encodings should be part of out-of-the box Python, so no DLL is missing after you added python38.dll to the install directory.

I suggest you try to install a newer version of Python, is possible, and then try running Kactus2 again.

0tti, 2021-09-13:
I can not update the python version due to other compatiblity issues. 
Wouldn't it be better when the Kactus2 tool bundles the python version and modules it needs to run independent? It produce always confilicts to rely on a user installation. The user will  have newer or older versions which are maybe partly incompatible.
E.g. Modesim brings also it's own TCL interpreter to be on the save side.

g-mahieux, 2021-11-05:
I myself also have this issue, as well as various colleagues.

teuhom, 2021-11-30:
We are working on this.
In the meantime, you could install Python 3.8, that should make it work.

teuhom, 2021-12-07:
We have released a new version of the Windows installer which should address this issue.

0tti, 2021-12-07:
With 3.9.168 Kactus2 starts on my machine.

closed 2021-12-15 by teuhom