kactus2/kactus2dev #49: Component port export / import are not compatible. Ad-hoc import is always true.

TonyReinberger, 2022-02-09:
If I export the the ports as a csv file and them import it again, the Ad-hoc is always set to true. I tried editing the file to use 0 and 1 and it still imports as true.

TonyReinberger, 2022-02-09:
If you copy part of the Ports Ad-hoc column that's false and paste it into another part of the Ad-hoc column, it will invert the value. If you paste a value that's true then the value will be true.

TonyReinberger, 2022-02-09:
These should be separate issues but the Parameter export and import are not compatible when the value consists of a formula involving an ID. It is likely that the ID has not been created yet so a second pass is needed to resolve the name given that the order could be random. I'm sure the other formula columns are likewise affected. I'm not sure if the parameters in Verilog and VHDL need to be in the correct order of definition when generated, but I suspect they do.

epekkar, 2023-04-18:
Fixed in e7eb2952a364c9e6a52c6944cce88eb68cddc259

closed 2023-04-18 by epekkar