kactus2/kactus2dev #51: Navigating components is very slow with 100s of ports.

TonyReinberger, 2022-02-13:
Kactus2 3.9.171 64bit Windows

It's slow at importing a large number of ports. Once they are imported, it is very slow at getting to the portmaps for the bus interface. Perhaps the algorithms can be improved because I think it gets exponentially slower as you add ports. It does not seem to be multithreaded to use more cores. The task can take minutes to 10 minutes or more (still waiting). It was previously slow with 300 ports and I'm not sure how slow it will be with close to 1000 ports so now I have lots of time to write this issue.

Great program though and I'm still trying to learn it. I'm hoping I can use it to manage a top level design. It has a lot of AMBA registers so it has lots of field signals going to the AMBA block.

Cheers,

Tony

TonyReinberger, 2022-02-14:
I don't have an exact time but it took about 30 minutes to open the port map tab with 905 ports. I then auto connected the bus and it took 2 hours. The names were not identical because the component ports had "_in" and "_out" suffixes while the bus signals did not. It did complete though, so don't force quit the program because it's not responding. Be patient, it comes back.

epekkar, 2022-02-17:
Hi Tony,
I'll have a look at this later with performance evaluation tools, but my guess is that the the cause is evaluating the values (port bounds etc.) in the table. Each row has 5 items that may contain a function and running the evaluation and validation of the rows is just too much.

TonyReinberger, 2022-02-17:
> Hi Tony,
> I'll have a look at this later with performance evaluation tools, but my guess is that the the cause is evaluating the values (port bounds etc.) in the table. Each row has 5 items that may contain a function and running the evaluation and validation of the rows is just too much.

Thanks, I've partially gotten around the problem by scripting the portMaps and placing them directly in the xml file for the component. This way I could do the top HW design connections.

TonyReinberger, 2022-03-22:
I think parameters make the design a lot slower as well. I thought I would add that note since the high port count may be only part of the slowdown. If the ports are using parameters as the width, minutes might turn into hours.