kactus2/kactus2dev #55: Modern linux can't load executables as libraries

olofk, 2022-03-15:
I'm getting the following error when using the Python API

```
Scanning library. Please wait...
Ready.
Validating items. Please wait...
========== Library integrity check complete ==========
Total library object count: 6
Total file count in the library: 0
Ready.
Updating library view. Please wait...
Ready.
Python 3.8.10 (default, Nov 26 2021, 20:14:08) 
[GCC 9.3.0]
Could not import Kactus2 PythonAPI.
Traceback (most recent call last):
  File "/mnt/gentoo/home/olof/projects/kactus2/kactus2dev/PythonAPI/pythonAPI.py", line 15, in <module>
    import _pythonAPI
ImportError: libKactus2.so: cannot dynamically load position-independent executable
```

After some digging it seems like libKactus2.so is just a renamed kactus2 binary and modern Linux doesn't allow loading executables as binaries.

A quick (and probably not recommended) workaround that I found [here](https://lief-project.github.io/doc/latest/tutorials/08_elf_bin2lib.html) is to pretend it's not an executable with a small python hack

```
import lief
path = "libKactus2.so"
bin_ = lief.parse(path)
bin_[lief.ELF.DYNAMIC_TAGS.FLAGS_1].remove(lief.ELF.DYNAMIC_FLAGS_1.PIE)
bin_.write(path)
```
The proper solution would be to separate the library from the executable

epekkar, 2022-03-15:
This is the second time I see this problem and in this context it makes much more sense. I will separate the library and the executable.

epekkar, 2022-05-11:
Ok, I finally got around to fixing this one. I'm running final checks, but it should now work as intended with a new library in place.

olofk, 2022-05-11:
Cool. I'll give it a try when I can find the time

olofk, 2022-05-12:
Works perfectly fine!

closed 2022-05-12 by olofk

epekkar, 2022-05-12:
Excellent. Thanks for confirming that, Olof.