kactus2/kactus2dev #56: Python interface library initialization

olofk, 2022-03-18:
When I run `kactus2 -c` I'm greeted with some statistics about the IP-XACT library.

However, if I run python directly and run e.g.
```
import pythonAPI
api = pythonAPI.PythonAPI()
api.vlnvExistsInLibrary(....)
```

I will get a segmentation fault. I guess it's because the library hasn't been initialized properly and I suspect I need to run setupLibrary somehow but haven't figured out what I need to do

epekkar, 2022-03-21:
For now the IP-XACT library is only usable if Kactus2 is running (Kactus2 takes care of the library setup). I'll check, if there is a workaround with setupLibrary that is functioning now. Having both options takes some work, but the initialization should be automatic either way.

epekkar, 2022-05-30:
Should be fixed now. You can import pythonAPI module and use it without starting Kactus2 first.

closed 2022-06-13 by epekkar