kactus2/kactus2dev #58: Verilog generation adds VHDL port type.

TonyReinberger, 2022-03-22:
I've imported a VHLD component which adds the type to the ports.
```
library IEEE;
use IEEE.STD_LOGIC_1164.all;
use IEEE.STD_LOGIC_arith.all;
use IEEE.STD_LOGIC_unsigned.all;
                            
entity DIG_TOP is
  port (
    clk_in  : in std_logic;
    rstb_in : in std_logic
  );
end DIG_TOP;
```
[image]

However, when this is then generated as a Verilog file, the type is included which is wrong.

[image]

Anytime the component is re-imported, the type needs to be manually deleted.

epekkar, 2022-03-23:
Hi Tony,
This is expected behavior. The generated port type is determined by the view selected for generation. If you double-click the type cell, you will see the type definitions can be made for each view (empty view name in the table is regarded as the default). If there is only one view, the only available type (std_logic) will be used.

closed 2022-03-23 by epekkar

TonyReinberger, 2022-03-23:
Hello,

If I only have a Verilog view but I'm importing a VHDL top file then it seems I have no choice but to manually delete the type. If I make a VHDL view and then import, how can you select the view so the types are applied only to that view?

The situation is that a person is doing a component design in VHDL but the rest of the hardware is Verilog. When they make port changes, the VHDL top is imported, interfaces are updated and then the hardware verilog is generated.

Regards,

Tony

TonyReinberger, 2022-03-23:
I should add the verilog component view is generated as an empty module for partial synthesis.

epekkar, 2022-03-24:
I think the correct way here is to define two views: one for Verilog, one for VHDL. You could even name them to match the use to something like generated_verilog and imported_vhdl. Then after the import you need to specify that the imported types only apply to VHDL. This is set in the ports editor as shown in the screenshot below. The trick here is to right-click on the type to add the view reference i.e. to which view(s) the type applies. 
[image]

Now you should be able to run the generation for view generated_verilog without the std_logic type.

TonyReinberger, 2022-03-24:
Thanks for the information. That's a lot of work with 505 ports so I would script the changes to the XML directly. Instead, I think I'll remember to select the column and delete the types after an import.