kactus2/kactus2dev #63: cannot use executable file libKactus2.so as input to a link

vowstar, 2022-04-14:
When compile latest kactus2(git hash: a1d144eb6777a1a7dc88dc69637a0a8c7b94edcb) form source on linux (latest archlinux, gentoo, debian sid), errors will occur during compilation.

```bash
 cannot use executable file '/home/vowstar/Projects/2022/kactus2dev/PythonAPI/../executable/libKactus2.so' as input to a link
```

log:

```bash
./configure
make
...
...
rm -f libPythonAPI.so.1.0.0 libPythonAPI.so libPythonAPI.so.1 libPythonAPI.so.1.0
g++ -Wl,-O1 -shared -Wl,-soname,libPythonAPI.so.1 -o libPythonAPI.so.1.0.0 release/PythonAPI.o release/ComponentAndInstantiationsParameterFinder.o release/ComponentParameterFinder.o release/ExpressionFormatter.o release/IPXactSystemVerilogParser.o release/ListParameterFinder.o release/ParameterCache.o release/SystemVerilogExpressionParser.o release/AddressBlockExpressionsGatherer.o release/FieldExpressionsGatherer.o release/MemoryMapExpressionsGatherer.o release/MemoryRemapExpressionGatherer.o release/RegisterExpressionsGatherer.o release/RegisterFileExpressionsGatherer.o release/librarypathvalidator.o release/VersionHelper.o release/PythonAPI_wrap.o release/moc_ParameterCache.o release/moc_librarypathvalidator.o  -Lx64/executable -L/home/vowstar/Projects/2022/kactus2dev/PythonAPI/../executable -lIPXACTmodels -lKactus2 /usr/lib64/libQt5Widgets.so /usr/lib64/libQt5Gui.so /usr/lib64/libQt5Xml.so /usr/lib64/libQt5Core.so -lGL -pthread    
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: cannot use executable file '/home/vowstar/Projects/2022/kactus2dev/PythonAPI/../executable/libKactus2.so' as input to a link
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:143: libPythonAPI.so.1.0.0] Error 1
make[1]: Leaving directory '/home/vowstar/Projects/2022/kactus2dev/PythonAPI'
make: *** [Makefile:111: sub-PythonAPI-make_first] Error 2

```

This may be related to issue https://github.com/kactus2/kactus2dev/issues/55

vowstar, 2022-04-14:
Reference: https://stackoverflow.com/questions/68622938/new-versions-of-ld-cannot-take-elf-files-as-input-to-link

epekkar, 2022-04-14:
Duplicate of #55. We are working on fixing this.

closed 2022-04-14 by epekkar