kactus2/kactus2dev #66: Unreasonable Python version dependency

ktbarrett, 2022-08-30:
The README states that a minimum Python version of 3.10 is required to build Kactus2 on Linux. This is a totally unreasonable expectation for a tool meant to be used by professionals, which are mostly stuck with whatever version of Python comes with the latest version of RHEL (3.6 on RHEL8).

If the Python API is an optional component, or could be made an optional component, and slackening the Python version requirement is not possible, please update the build scripts to make building the Python API optional.

epekkar, 2022-08-31:
Hi ktbarrett,
The current build should work with at least Python version 3.8. If you want to remove PythonAPI from the build completely, you can do this by commenting out PythonAPI under the SUBDIRS section in Kactus2_Solution.pro. At the time of writing this is line 11.

ktbarrett, 2022-08-31:
I'll try that. Would it be acceptable to add a flag (to configure probably?) specifically to turn this on/off easily. This flag can then be documented.

andrewandrepowell, 2022-09-08:
I have confirmed Python 3.8 does work after making the suggest change in https://github.com/kactus2/kactus2dev/issues/66#issuecomment-1232618222.