kactus2/kactus2dev #77: Unable to build the latest commit on the master branch

AmeyaVS, 2023-03-18:
It seems the latest commit: e6d55fb8614623c765677f9f0b158404e74f6eb1 has completely broken the tool.
GUI keeps crashing, unable to keep the GUI open more than few seconds on Linux.

teuhom, 2023-03-22:
Hi.
I have tested this build on Ubuntu 22.04, and it seems to be working fine, with the exception of the new generator plugin.
Could you give more details about the steps that lead to the crashes? For example if it crashes on startup or having a component or design open.

AmeyaVS, 2023-03-22:
I am using Fedora 37 as a base system.
The crash is usually happening during startup itself.

AmeyaVS, 2023-03-23:
After cleaning up the build directory extensively, It seems I am not able to reproduce the underlying issue.
Closing the issue for now.

closed 2023-03-23 by AmeyaVS