kactus2/kactus2dev #82: How do I generate svd files

Yaozzheng, 2023-08-17:
hello,

I want to use svd generator to generate an svd file for the sample project 'ipxactexamplelib', the manual has no instructions, and the video tutorial is out of date.

Here's how I do it:
1. Import the sample project.
2. Open --> tut.fi -> cpu.structure --> cpu.example  -->Click OPEN HW design -->SVD generator.

After I click svd generator, there is no drop-down option for file set in the prompt box, and an error message is displayed indicating that can not find cpu routes. Do I need to manually add file set or improve other information?

epekkar, 2023-08-17:
Hi Yaozzheng,

The problem with the example you choose is that it is missing the CPU element just like you noticed in the error message. The SVD generator expects to find a CPU with an attached Address Space in the same component and other components with memory maps.

The file set selection is a bit different. The file set you select (or disable) in the dialog determines the target file set in which the generated files are added into. The drop-down menu will suggest existing file sets or alternatively filling in a new name will create a new file set. By clicking on "Add files to file set" you can direct the generator to create the files without adding the files in the component file sets.

Yaozzheng, 2023-08-17:
Hi,epekkar,

Thank you very much for your reply,

In the latest version of the example in the project "https://github.com/kactus2/ipxactexamplelib", when I click on 'tut.fi/cpu. structure/cpu_example' - Open HW design, I can see that there are cpu core, buses and some peripherals, and by clicking Open Memory design I could see the address map of the core and peripherals.

Is This component 'cpu_example' a top-level file? I don't know how to add cpu information and memory information to this component.

epekkar, 2023-08-17:
The cpu_example is not really a good example for the SVD generation. Although, it has a component representing a cpu core, that component does not have a CPU element. The CPUs are added in the Component editor in the CPUs section on the left. SVD generator requires at least one CPU to be present. The Memory Desiger on the other hand does not require a CPU, but only address space(s) and memory map(s).

CalamaroS, 2023-11-30:
Hi @epekkar I'm currently having trouble to obtain an SVD starting from the memory maps of my mcu's peripherals, but I'm getting this output: 
'Running SVD Generator 0.2.
Could not find CPU routes' 
Is there a tutorial for SVD generation or can you list the actions required to obtain such a result? Thank you very much.