kactus2/kactus2dev #83: Missing Fields in Memory Maps Visualization

pascal-creonic, 2023-08-23:
In some cases fields are not showing in the memory maps visualization.
What I have found:
1. If only a single field is present, it is not showing up, no matter what.
1. If two fields are present, which do not occupy the full register, they are showing up
1. If two fields are present, which do occupy the full register, they are not showing up. However, for more than two fields occupying the full register, they are showing up
1. If one of two fields' isPresent value is set to false, no fields are shown. However, if more than two fields are defined of which only one is present, it is showing up

[demo.1.txt](https://github.com/kactus2/kactus2dev/files/12416701/demo.1.txt) contains the minimal example producing the memory maps visualization shown below.
[image]

Version: kactus2 3.12 built on Ubuntu 22.04.3 LTS (GNU/Linux 5.15.0-75-generic x86_64) from the release tar-package

epekkar, 2023-11-28:
Thanks for reporting this one. We will investigate it in more depth, but so far we found out that the visualization only fails on Linux.

hagantsa, 2024-01-17:
Fixed in 43096fd1e837ac30bafcb622438551ae337afa37.

closed 2024-01-17 by hagantsa