kactus2/kactus2dev #87: Shutting down unexpectedly

DenizzzGuzell, 2023-11-22:
Hi, In Kactus2 version 3.12.0 shuts down unexpectedly in error cases. I am currently using Kactus2 on W11 machine and downloaded from SourceForge. I think error cases responsible for this bug. For example if you do not select abstraction definition and try to add row to port map it shuts down. There are more like that. It should not shuts itself immediately when there are mistaken requests are occurs.

epekkar, 2023-11-22:
Hi Deniz,

Thank you for reporting the issue. The port map issue is known to us and will be fixed for version 3.13. Would you kindly elaborate on other cases where the shutdown happens?

DenizzzGuzell, 2023-11-22:
When creating HW component, imported source files can have declarations like "parameter logic [NrHarts-1:0] SelectableHarts  = {NrHarts{1'b1}},". It generates instant error. By the way, I am currently working on openhwgroup core-v-mcu project to build it in kactus2. These types of declarations classified as error and after that when I try to generate component, Kactus2 immediatly shuts itself. Everytime importing file which has a bit complex parameters or ports declaration as given above shuts programme without saving anything. That why when I import these type of files I delete ports from rtl file which is not correct method. After generating component without ports, I added ports manually again. I am not currently remembering all of the declarations which caused problem, but one of them is given above. Under this issue I will share other types of declarations which are cause problem on kactus2, if I remember.  I am sharing screenshot to better understanding: 
[image]
This will shut itself when finish or next/finish selected.

DenizzzGuzell, 2023-11-23:
Another shutting down occured right now. When defining file sets if you right click undefined files and select define a folder after that Kactus2 shuts itself.

DenizzzGuzell, 2024-01-08:
Tested it solved i think.