kactus2/kactus2dev #93: Supporting alternate register

DenizzzGuzell, 2023-12-04:
Hi, I could not find any option to design memory using alternate registers such as uart receiver address 0 read only and uart transmitter register address 0 write only. It throws error as ` Registers rcv_buf and tr_buf overlap within address block UART`. Does Kactus2 supports alternate register description?

epekkar, 2023-12-05:
Hi Deniz,

Currently Kactus2 does not support alternate registers.

DenizzzGuzell, 2024-01-08:
Hi epekkar,

Do you have any plan to add alternate registers to Kactus2 because of high necessity and usage of alternate registers.

hagantsa, 2024-01-17:
Hi Deniz,

Alternate registers are something we probably will want to add to Kactus2, but there is currently no timeline set for it.

DenizzzGuzell, 2024-01-18:
Hello, 

Thank you for answer.

closed 2024-01-18 by DenizzzGuzell