kactus2/kactus2dev #95: Qt6 requires purchase by commercial users

lmg260a, 2023-12-20:
Hi,
I'd really like to try Kactus2 and show the value of IP-XACT at my company, and (hopefully) persuade them to do a couple of test projects using Kactus2.

We do all our work on linux.
However, Kactus2 for linux requires Qt6 which requires purchasing a license.  Which means I have to get Legal department involved, management approval to purchase, etc. etc.

I.e. demonstrating Kactus2 is dead in the water.

I hope that there will be some way to resume support for Qt5, which doesn't require a license on linux.

vowstar, 2023-12-21:
> ``Kactus2 for linux requires Qt6 which requires purchasing a license.``

Based on the Qt 6 licensing documentation, using Qt 6 for open-source projects like Kactus2 does not require a commercial license if the project complies with the GNU LGPL or GPL terms. The misconception that Qt 6 necessitates a license purchase for Linux is not accurate. Qt allows free use in open-source projects under certain open-source licenses. For more detailed information, you can refer to the Qt 6 licensing page: https://doc.qt.io/qt-6/licensing.html

Most modern versions of Linux distributions like Debian, Fedora, Arch Linux, and Gentoo already include Qt 6 in their repositories, allowing for straightforward installation. Furthermore, for older versions of these distributions, it's possible to compile and install Qt 6. This availability in standard Linux repositories further supports the accessibility of Qt 6 for open-source projects without the need for a commercial license.

hagantsa, 2024-01-04:
Hi,

As @vowstar already said, a commercial Qt6 license is not required to build Kactus2 for linux. As for Qt5, we currently have no plans to resume support for it.

closed 2024-01-04 by hagantsa