kactus2/kactus2dev #97: make first execution Error

Ryuji-Yatahagane, 2024-01-12:
[image]

Hello everyone, 
I´m trying to compile and i got this error at the firs $make command.

It´s a known error ? 

Greatings,

hagantsa, 2024-01-16:
Hi,

This error is most likely caused by an invalid `PYTHON_CONFIG` variable in `.qmake.conf`. The `PYTHON_CONFIG` variable should match the installed Python version. For example, for Python 3.11 it should be set to `python3.11-config`. Please make sure to follow the build instructions closely.

closed 2024-01-16 by hagantsa

Ryuji-Yatahagane, 2024-01-17:
Hello,
Thanks u so much for ur suggestion 
solves the problem.