kactus2/kactus2dev #116: Discussion Forum query

AmeyaVS, 2024-08-05:
I have a query regarding any discussion forum for the development activities other than github issues.

I have been using the tool for quite sometime, but I always face issues when pulling latest changes from project source updates.
I have created scripts to setup a clean development environment every time I update the sources but this doesn't scale very well when developing plugins as I am wasting a lot of compute cycles on rebuilding the whole project every time and also installation errors where the install targets sometimes doesn't pick the latest build shared libraries or is using older version on the install path.

Is there a discussion forum where I can discuss build issues and to better understand project source organization from the maintainers other than GitHub issues? something along the lines of mailing list, etc.?

Regards,
Ameya

hagantsa, 2024-08-12:
Hi Ameya,

Currently we do not have any sort of forum or mailing list apart from this issues page. We could look into opening the github discussions page (or some other platform) at some point, if there is interest in the community.

In the meantime, can you describe the update problems more closely? I don't quite understand what you mean with having to rebuild the entire project when pulling changes. Do you mean that every time you build your plugins you also have to rebuild the entire project? Also, can you describe in which situations the install fails?

AmeyaVS, 2024-08-19:
Hello @hagantsa ,

Sorry for the late response.
I would appreciate if some sort of searchable discussion forum would be more appropriate on asking questions related to the project build and configurations.

I am currently building the Kactus2 tool in a limited privilege environment i.e. no admin/root privilege to install the tool in the system. I have found inconsistent plugins builds when developing extensions for Kactus2.
I would add/modify existing extensions to add functionality build the project/plugin and then install.
It would seem the install steps sometimes skip installing updated plugin/extension binaries to target/install paths.

I would need to frequently do a clean/build and also cleanup the install paths so I can ensure the plugins are installed correctly.

hagantsa, 2024-09-11:
Hello,

Sorry for the delayed response. When I have time I will try to replicate the issue you are encountering with building and installing plugins. Local user kactus2 installs and custom plugins on linux are not that thoroughly tested, as we primarily focus on our windows builds.